esp-rs/esp-backtrace #4: Prettier panic formatting?

har7an, 2022-08-17:
Hello,

started using this crate today after it got pulled in by the no_std application template. I see that currently it just debug-prints the `PanicInfo`, but would you accept a PR that adds some prettier formatting around it?

In a minimal esp32 code that prints "Hello" and then panics, the serial output is pretty crowded imo:

```
load:0x3fff0054,len:4800
load:0x40078000 [_stack_end_cpu0:??:??],len:17448
load:0x4007c428 [_stack_end_cpu0:??:??],len:4840
entry 0x4007c6a0 [_stack_end_cpu0:??:??]
Hello
PanicInfo { payload: Any { .. }, message: Some(Help me, I'm dying!), location: Location { file: "src/main.rs", line: 30, col: 9 }, can_unwind: true }
0x400d1bb8 [half_duplex_uart::__xtensa_lx_rt_main:/var/home/ahartmann/repos/mt/playground/half-duplex-uart/src/main.rs:30]
0x400d5816 [Reset:/usr/local/share/cargo/registry/src/github.com-1ecc6299db9ec823/xtensa-lx-rt-0.13.0/src/lib.rs:77]
0x400d29af [ESP32Reset:/usr/local/share/cargo/git/checkouts/esp-hal-42ec44e8c6943228/6e037b0/esp32-hal/src/lib.rs:78]
```

I was thinking about maybe something like this:

```
load:0x3fff0054,len:4800
load:0x40078000 [_stack_end_cpu0:??:??],len:17448
load:0x4007c428 [_stack_end_cpu0:??:??],len:4840
entry 0x4007c6a0 [_stack_end_cpu0:??:??]
Hello

!! A panic occured at "src/main.rs", line 30, column 9

  Message: Help me, I'm dying!
  Payload: ??
Backtrace:

0x400d1bb8 [half_duplex_uart::__xtensa_lx_rt_main:/var/home/ahartmann/repos/mt/playground/half-duplex-uart/src/main.rs:30]
0x400d5816 [Reset:/usr/local/share/cargo/registry/src/github.com-1ecc6299db9ec823/xtensa-lx-rt-0.13.0/src/lib.rs:77]
0x400d29af [ESP32Reset:/usr/local/share/cargo/git/checkouts/esp-hal-42ec44e8c6943228/6e037b0/esp32-hal/src/lib.rs:78]
```

Or in accordance with the default rust panic handler:

```
!! A panic occured at src/main.rs:30:9
```

bjoernQ, 2022-08-17:
Sure, currently it only debug prints the `PanicInfo` - that could be improved

har7an, 2022-08-17:
Hmm, it seems that obtaining the panics message is an unstable feature. I've opened a draft PR #5, but that's not really an improvement unless the feature is stabilized.

bjoernQ, 2022-08-17:
ah yes ... probably the reason why I opted for using that ugly debug print

jessebraham, 2022-08-19:
Closed by #5

closed 2022-08-19 by jessebraham