esp-rs/esp-backtrace #9: Add a feature for logging panics with defmt

TheButlah, 2022-09-19:
In line with the `print-uart` and `print-rtt` features, perhaps a `print-defmt` would make sense?

I am happy to submit a PR for this, however I have a question. The current implementation calls several `println!()` functions. In most defmt log printers, the contents of the log statement gets printed along with the location in the source code that the log statement was called.

Would this mean that we would have only one `error!()` invocation in the panic_handler? Or should we still call `error!()` each time we call `println()`

bjoernQ, 2022-09-19:
I'm not too familiar with _defmt_ to be honest. Probably `error!()` is more like `dbg!` from the std-lib, then?
Sounds like you probably only want one invocation of `error!` in your implementation

By "should I add this functionality to `esp-println`" you mean printing the source code location? I don't think we should do that for `println` but if you like you can add a `dbg` macro just like https://github.com/rust-lang/rust/blob/efa717bc2d82c19c6957f362a936a6f95169d138/library/std/src/macros.rs#L339-L361 to _esp-println_

Otherwise, happy to receive a PR for this

TheButlah, 2022-09-19:
> By "should I add this functionality to esp-println" you mean printing the source code location?

No, I meant should I add a `print-defmt` feature to `esp-println`. I'm not sure if its a good match because `defmt` introduces its own formatting. See this example:
[image]

jessebraham, 2023-09-29:
I'm just starting to learn `defmt` myself, but `esp-println` at least has support for it now.

bjoernQ, 2024-01-15:
I guess this is covered by #43

closed 2024-01-15 by bjoernQ