esp-rs/esp-backtrace #32: file not found for module `arch`

imizao, 2023-06-13:
Hi, When I run cargo run, it reports this error！

```
error[E0583]: file not found for module `arch`
 --> /Users/xxx/.cargo/registry/src/mirrors.ustc.edu.cn-61ef6e0cd06fb9b8/esp-backtrace-0.7.0/src/lib.rs:8:1
  |
8 | pub mod arch;
  | ^^^^^^^^^^^^^
  |
  = help: to create the module `arch`, create file "/Users/xxx/.cargo/registry/src/mirrors.ustc.edu.cn-61ef6e0cd06fb9b8/esp-backtrace-0.7.0/src/arch.rs" or "/Users/xxx/.cargo/registry/src/mirrors.ustc.edu.cn-61ef6e0cd06fb9b8/esp-backtrace-0.7.0/src/arch/mod.rs"

error[E0425]: cannot find function `backtrace` in module `crate::arch`
  --> /Users/xxx/.cargo/registry/src/mirrors.ustc.edu.cn-61ef6e0cd06fb9b8/esp-backtrace-0.7.0/src/lib.rs:31:34
   |
31 |     let backtrace = crate::arch::backtrace();
   |                                  ^^^^^^^^^ not found in `crate::arch`

Some errors have detailed explanations: E0425, E0583.
For more information about an error, try `rustc --explain E0425`.
error: could not compile `esp-backtrace` (lib) due to 2 previous errors

```

MabezDev, 2023-06-13:
This crate only supports Xtensa or riscv, ensure you are building for the correct target.

closed 2023-06-13 by MabezDev

patrickelectric, 2023-11-02:
I'm having the same problem:

```
error[E0583]: file not found for module `arch`
 --> /home/patrick/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-backtrace-0.9.0/src/lib.rs:8:1
  |
8 | pub mod arch;
  | ^^^^^^^^^^^^^
  |
  = help: to create the module `arch`, create file "/home/patrick/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-backtrace-0.9.0/src/arch.rs" or "/home/patrick/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-backtrace-0.9.0/src/arch/mod.rs"

error[E0425]: cannot find function `backtrace` in module `crate::arch`
  --> /home/patrick/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-backtrace-0.9.0/src/lib.rs:31:34
   |
31 |     let backtrace = crate::arch::backtrace();
   |                                  ^^^^^^^^^ not found in `crate::arch`

Some errors have detailed explanations: E0425, E0583.
For more information about an error, try `rustc --explain E0425`.
error: could not compile `esp-backtrace` (lib) due to 2 previous errors

Caused by:
  process didn't exit successfully: `/home/patrick/.rustup/toolchains/esp/bin/rustc --crate-name esp_backtrace --edition=2021 /home/patrick/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-backtrace-0.9.0/src/lib.rs --error-format=json --json=diagnostic-rendered-ansi,artifacts,future-incompat --diagnostic-width=273 --crate-type lib --emit=dep-info,metadata,link -C opt-level=s -C embed-bitcode=no -C debuginfo=2 -C debug-assertions=on --cfg 'feature="esp-println"' --cfg 'feature="esp32s2"' --cfg 'feature="exception-handler"' --cfg 'feature="panic-handler"' --cfg 'feature="print-uart"' -C metadata=386bc5d608c0c2a8 -C extra-filename=-386bc5d608c0c2a8 --out-dir /home/patrick/git/patrick/weather-esp32-s2/target/debug/deps -L dependency=/home/patrick/git/patrick/weather-esp32-s2/target/debug/deps --extern esp_println=/home/patrick/git/patrick/weather-esp32-s2/target/debug/deps/libesp_println-3cf1b2e2e6af783f.rmeta --cap-lints allow` (exit status: 1)
```

bjoernQ, 2023-11-03:
Would you mind sharing the whole project? We select the `arch` by the target architecture (https://github.com/esp-rs/esp-backtrace/blob/db2bb85bc9c24800287912efe305606a9f9ebb2a/src/lib.rs#L6-L8) so it seems you are building for something else

patrickelectric, 2023-11-03:
You can check it here: https://github.com/patrickelectric/weather

bjoernQ, 2023-11-03:
Seems like you are missing the `.cargo/config.toml` file there which sets the default target. See https://github.com/esp-rs/esp-hal/blob/main/esp32s2-hal/.cargo/config.toml