esp-rs/esp-backtrace #37: Option to reset instead of halting

bugadani, 2023-10-12:
Halting is fine for development, but I can't really expect end-users to grab a pair of tweezers and short a pair of pads in case their device panics. At least not in every one of my projects :)

bjoernQ, 2023-10-13:
True - until now this is mainly meant to help during development. Not really thought about the production use case.

I could also think about some additional features like persisting the backtrace and panic/exception information - then the application could optionally send this to a server on next boot

bugadani, 2023-10-13:
Persisting is a good idea. To spin it further, I think it should be opt-in, with a user-provided buffer. A bit tricky to use because the buffer needs to be placed in an uninitialized section, but one can then choose how much memory one can sacrifice for something that, in an ideal world would never be used :)